Manuel83/craftbeerpi3 #3: Cannot get auto to start heating

Blom1974, 2017-06-11:
Brewstep is started and kettle is in auto mode but heater does not start.

[image]

Blom1974, 2017-06-11:
It was some problem with actuator not saved on kettle.

Manuel83, 2017-06-11:
Can you post a picture of your hardware settings, please. Including a picture of your kettle config

Blom1974, 2017-06-12:
I think this was an human error. I saved a kettle with an actuator for heat. But then i deleted and and made a new actuator which in turn made the kettle actuator to be empty.

It might be good to have som form of check. Not not really sure of how though.
Warning when deleting actuator which is in use? "this actuator is used in kettle x and kettle y.

closed 2017-06-12 by Blom1974