Manuel83/craftbeerpi3 #8: Enhancement Adding Cycle lenght to GPIOPWM

Blom1974, 2017-06-12:
It is nice to have an cycle lenght option on the GPIOPWM. For example an SSR with zero crossing start might not work good with very short periods. Then you might for example set cycle lenght to 10 sec and then modulate the time 0-100% of that.

hjarupBrewDK, 2017-06-12:
I have the same observation and measurement. I my view a sensible period is >6sec. 
This will also make it usefull if using mechanical relays.

(see attached Picture of Measurement on PIO from RPI)
[image]

Blom1974, 2017-06-12:
Nice explanation!

seesser, 2017-06-16:
When I made invertPWM I built it off the GPIOPWM.  When you setup the Actor under Duty Cycle is actually the frequency Cycle.  So just change the duty cycle to what you want the frequency cycle to be.  Power under the main screen is then the Duty Cycle or percent of the cycle it is on.  I will add the Duty Cycle name as a bug report.

closed 2018-03-17 by Blom1974