Manuel83/craftbeerpi3 #15: Auto Mode does not work when Using Fahrenheit

seesser, 2017-06-14:
Auto mash mode does not work when using US measurements.  If setup in Parameter as F for unit it will not control or start the timers.   Current workaround is to convert all measurement to Celsius and change Parameter to C also before running in auto mode.

alexpalmer, 2017-06-17:
Seconded. C works great, but F does not

IndyJoeA, 2017-06-18:
I am writing a plugin and it seems like the sensor value is being stored as a float data type for C, but as a string for F.  Maybe this is why the other routines are not working if using F.

Think it might happen here:
[https://github.com/Manuel83/craftbeerpi3/blob/master/modules/base_plugins/one_wire/__init__.py](https://github.com/Manuel83/craftbeerpi3/blob/master/modules/base_plugins/one_wire/__init__.py)
```python
    def read(self):
        if self.get_config_parameter("unit", "C") == "C":
            self.data_received(self.t.value)
        else:
            self.data_received(format(9.0 / 5.0 * self.t.value + 32, '.2f'))
```

alexpalmer, 2017-06-18:
Correct! 
Updated to this (round,2)
```
def read(self):
        if self.get_config_parameter("unit", "C") == "C":
            self.data_received(self.t.value)
        else:
            self.data_received(round(9.0 / 5.0 * self.t.value + 32, 2))
```

And it works now... Not sure how to submit the fix!

@IndyJoeA: great work on the Blynk plugin - I was using that prior to craftbeerPi

IndyJoeA, 2017-06-18:
Thanks :)  I noticed it because the data type was changing when I sent the sensor values to Blynk, which was causing weird formatting issues.  I put a check in my code to look for type str or type float and adjust accordingly.

IndyJoeA, 2017-06-18:
While this isn't related to this issue, but in the same spot - I would think both readings should be rounded to 2, whether in C or F mode, for consistency.  Would look better on the dashboard too!

seesser, 2017-06-22:
Thank you all for the great work.  Issue has been fixed.

closed 2017-06-22 by seesser