Manuel83/craftbeerpi3 #16: Auto mode does not work in Fermentation

helibrewer, 2017-06-14:
Manual switch operation functions normally.
Auto mode fahrenheit does not work for cooling
Auto mode fahrenheit does not work for heating
Auto mode celsius does not work for cooling
Auto mode celsius does not work for heating

Tried on GPIO pins 17 and 22 (both non-pwm)
Used GPIOSimple to set up switch actor
Deleted Craftbeerpi and completely reinstalled -- no change
Auto mode celsius heating works for Brewing

Verified hardware by using my Craftbeerpi 2.2 SD card and auto mode worked for Fermentation

swimIan, 2017-06-15:
I am expierencing something similiar. Heating work on Celsius but not Fahrenheit. I did not test cooling.

OliverE79, 2017-06-17:
I set up GPIO 18 as cooler (switches fridge by ssr).
The GPIO is never activated in auto mode, neither in hysteresis nor in pid mode. I am not sure if I configured  it correctly but I guess even with non optimal pid settings it should have switched at some point.

When adding a fermentation step the direction is displayed as "Heater", no matter if the temperature is below or above the setpoint. when I add a dummy actor and configure it as heater the direction seems to match - it starts to indicate "cool" also. But the GPIO is not activated.

helibrewer, 2017-06-22:
I noticed this error in the log after selecting the "auto" button on the fermenter:
ERROR:modules.app_config:Exception in function heater_off. Return default None

carlallen, 2017-06-23:
If you are using Fahrenheit and it's not working, that should be fixed by this PR https://github.com/Manuel83/craftbeerpi3/pull/24

Try again after updating.

helibrewer, 2017-06-23:
In modules.modes.core.controller.py, the following lines look like errors. they show f.heater and I believe they should be f.cooler

Lines 132 and 144

helibrewer, 2017-06-28:
Working after latest pulls

closed 2017-06-28 by helibrewer

got2brew, 2018-04-03:
I had the same issue on a single fermenter...  Found the issue.  I had accidentally erased the '0' in the heater off section. This made the automate not work. Once I put the zero back in the box, the auto mode worked again. (even though I was not using a heater)