Manuel83/craftbeerpi3 #22: Cannot download/view logs through System Menu

helibrewer, 2017-06-17:
When clicking the green "download" button I receive "404 Not Found" (http://192.168.1.50:5000/api/logs/download/app.log")

The red "trash" icon does delete the respective log file

JamFfm, 2017-06-17:
same to me

Manuel83, 2017-06-18:
I will check this

helibrewer, 2017-06-18:
Yes, app.log does exist in the logs folder

carlallen, 2017-06-19:
@helibrewer try restarting

helibrewer, 2017-06-19:
Still get the 404 error...and now when I use the back arrow on my browser after that error, I get the setup screen and lose my actors, fermenter, and kettle....the sensors remain.

peretti2004, 2017-07-09:
I also get the 404 error when I try to download logs.

carlallen, 2017-07-09:
I wonder if there is some kind of permissions issue. Can you do some things:
in the console in your craftbeerpi3 directory run:
`ps -ef | grep python`
and 
`ls -la logs`
then take a screen shot and post it here like this one:
[image]

helibrewer, 2017-07-09:
Here's mine:

root       715     1  0 Jun28 ?        01:52:43 /usr/bin/python -O /usr/share/wicd/daemon/wicd-daemon.py
root       725   715  0 Jun28 ?        00:43:49 /usr/bin/python -O /usr/share/wicd/daemon/monitor.py
root     15898     1  3 Jul07 ?        01:17:48 python /home/pi/craftbeerpi3/run.py
pi       19698 19645  0 07:37 pts/0    00:00:00 grep --color=auto python
pi@raspberrypi:~/craftbeerpi3 $ ls -la logs
total 144988
drwxr-xr-x 2 pi   pi        4096 Jun 29 16:37 .
drwxr-xr-x 8 pi   pi        4096 Jul  7 15:27 ..
-rw-r--r-- 1 root root 128516465 Jul  9 07:38 app.log
-rw-r--r-- 1 root root    821962 Jul  3 07:47 blynk.log
-rwxr-xr-x 1 pi   pi           0 Jun 28 07:48 empty
-rw-r--r-- 1 root root   4350835 Jul  9 07:38 fermenter_1.log
-rw-r--r-- 1 root root   4894230 Jul  9 07:38 sensor_1.log
-rw-r--r-- 1 root root   4892498 Jul  9 07:38 sensor_2.log
-rw-r--r-- 1 root root   4896148 Jul  9 07:38 sensor_3.log
pi@raspberrypi:~/craftbeerpi3 $

helibrewer, 2017-07-09:
[image]