Manuel83/craftbeerpi3 #25: Charts

brodiep, 2017-06-18:
Hi Manuel83, great software. 
The charts work fine until you do a git pull, then just blank charts
Nice work

Manuel83, 2017-06-18:
Can you try to delete all logs?

brodiep, 2017-06-18:
Sorry, did a clean install before I saw your response, back to where I started now, charts work, I will try some things and check the operation of the charts after every change.

helibrewer, 2017-06-18:
I have found that, depending on how much data you have, the charts can take 3+ minutes to load.

Ahoybakk, 2017-06-21:
I´m also having problems with the charts, they just show blank pages.

carlallen, 2017-06-21:
The problem with charts seems to be the quanity of data. The temperature is logged a lot more frequently than it was in 2.2 so it takes a lot longer to load which can leave you with a blank page for a long time.

IndyJoeA, 2017-06-22:
@carlallen Agreed.  I just started having this problem with blank charts today.  My Logs page won't load either, under the System menu.  I'm looking at the log files now and the app.log is 45 MB, sensor log is only 2 MB though.  Seems like the app.log file is logging debug information like continual sensor updates.  The amount of data could be vastly decreased.

jangevaare, 2017-06-22:
This same issue was encountered with Strangebrew Elsinore (other brewery software for Raspberry Pi). The solution was to only record temperatures when they changed past a certain threshold (rather than just every log the temperature every 5 seconds or something), as well as to limit the number of points plotted to something reasonable - a few thousand I believe.

IndyJoeA, 2017-06-22:
That must be it.  My sensor_x.log files are at 80,000 entries after just a week of testing.  Some kind of threshold value would help, plus the Python built in [RotatingFileHandler](https://docs.python.org/2/library/logging.handlers.html#rotatingfilehandler) function which can limit logs to a certain size (something like a few hundred KB for sensors perhaps), before it creates a new log or deletes old files.  So you would end up with sensor_1.log, sensor_1.log.1, sensor_1.log.2, etc.

helibrewer, 2017-06-22:
It might be better to write the data to the db, then you could select how much of the data you want to view. You could then export the db to a file, and purge it periodically

jangevaare, 2017-06-22:
To add to the idea of implementing a time series database, an automatic purging feature would be useful too - e.g. temperature data is retained for 60 days, or some user defined length of time. Also, plotting with some user defined resolution - say plotting every 5th, or 10th temperature record, and plotting with some user defined window - perhaps a couple hours, perhaps a couple weeks depending on the process.

carlallen, 2017-06-22:
I have a work-in-progress pull request that only logs temps if they change or if a minute has elapsed since the last time they changed. I'm thinking of making the time interval a config parameter. Thoughts?

https://github.com/carlallen/craftbeerpi3/pull/1

helibrewer, 2017-06-22:
carlallen, that sounds like a good solution. I have a basic understanding of Python and be able to help if anyone has simple stuff to do/test.

carlallen, 2017-06-22:
alright, I've got it in a functioning state. If anyone wants to give it a try and provide feedback feel free to do so. You will probably want to clear your logs after pulling my branch

carlallen, 2017-07-15:
I changed the way I was trying to fix this issue. My new fix now looks backwards at the log and prevents having 3 (or more) identical values in a row in the log. This should hopefully massively shrink logs.

capyy, 2017-12-04:
Hi @carlallen ,

Is this fix implemented ?
I started a fermentation on Friday night (the temp control works very well) but after 2 and a half days, my charts either takes 5+ minutes to load or load blank (just the charts' background lines) or do not load at all.

Thanks !!

SBassabe, 2018-01-29:
I've experienced the same problem during fermentation, in which the system needs to plot at least 7 consecutive days. I also noticed the the log captures 12 readings every minute, at this rate the charts module would need to plot 120,960 points (no wonder it takes more than 5 minutes to render a graph!!). 

As a workaround to this dilema I modified the method read_log_as_json() in the file /modules/logs/endpoints.py in order to create a copy of the original log file that captures only one reading per minute and sending this array to the plot module. This way the graph needs to plot only 10,000 for a 7 day fermentation period. The solution works well for long periods of time but it would be impractical for shorter periods, eg. mashing.

carlallen, 2018-01-29:
@SBassabe did you make a pull request?

SBassabe, 2018-01-30:
Hello Carl,
no I did not make a pull request :-( I just applied the fix locally to get
the chart going (need to keep an eye on my 23L Cooper Real Ale :-)
If you want to I could pass you the snippet of code.
Grazie, SBassabe

AlBrough, 2018-05-01:
This would be great if it were possible to write to a database (MariaDB etc) with a custom plotting interval, i.e. a 10 or 15 second resolution on each sensor, then for the graph to have a custom date and time ranges