Manuel83/craftbeerpi3 #26: Is there a way to send power to an actor

AHm74, 2017-06-18:
Is there any possibility to continuously send a desired power to the "actor" from tha "controller"?

Example, PID controller calculates 40% and I want the actor to set the PWM pulse width to 40%.

Maybe I'm missing a command but the only one I see for a "KettleController" is "heater_on" and "heater_off", would like an "set_power".

Manuel83, 2017-06-18:
Thanks for the hint. I just added a new API method.

https://github.com/Manuel83/craftbeerpi3/blob/master/modules/core/step.py

```
@cbpi.try_catch(None)
    def actor_power(self, id, power):
        self.api.actor_power(int(id), power)
```

In a KettleController you can call now:

`self.actor_power(1, 50)`

First parameter is the Actor Id and the second the power as percentage value

AHm74, 2017-06-18:
Thanks for the quick answer, however I don't understand how to use it.

I created
`@cbpi.controller
class PIDHendi(KettleController):`
and
`@cbpi.actor
class HendiControl(ActorBase):`

inside PIDHendi it's not possible to access 
`self.actor_power(1, 50)`

What am I missing? Shouldn't the base class `KettleController(ControllerBase, ActorController, SensorController): ` in craftbeerpi3/modules/core/controller.py contain a function for setting power?

Manuel83, 2017-06-18:
Sorry my fault. I added it only to the Steps and not the KettleControler.
Now it's available also for KettleController
You just need to extend KettleController

With the default heater id of the kettle call:`self.actor_power(50)`

With other actor id call: `self.actor_power(50, 1)`. The second parameter is optional

AHm74, 2017-06-18:
That did it!

closed 2017-06-18 by AHm74