Manuel83/craftbeerpi3 #27: Non visible marked sensor still visible after reboot

JamFfm, 2017-06-19:
My Temp sensor which is marked as hide still appears after reboot and is visible in Bewing. In Hardware dubbleclick on the checkbox (hide on dashboard) and press update and the sensor is again absend.

Manuel83, 2017-06-19:
I will check this

swimIan, 2017-06-27:
I'm having the same problem after the latest update but I haven't cleared the database yet.

IndyJoeA, 2017-06-27:
I'm on a brand new database (on a new Pi even) and I still have hidden sensors come back after a reboot.

swimIan, 2017-06-27:
I just cleared the database and still see hidden items on the dashboard too.

swimIan, 2017-07-24:
Any updates on this.

BrewChef, 2017-09-06:
Any news?  They dissapear when I hit hide, but after reboot - they are back.
Would be nice to fix this. I use the 7 inch Pi LCD, and with a 3 kettle setup, this makes the dahsboard rather messy.

KenCope, 2018-02-18:
I am having the same problems. This is 8 months after the original post. Does anyone know of a fix?

KenCope, 2018-03-20:
@Manuel83   Manuel83, I wish I knew enough to help fix this, but cannot.  Any update on it?  Thanks for all you do.

clearwaterbrewer, 2018-10-30:
Still broke....  should be easy fix...

Philusha1983, 2018-11-07:
I made a CSS hack to hide the required sensors:
please take a look at the video:
https://youtu.be/0_a43LK0PVg 

the code:
`    <style>

    /*hide on dashboard hack
    write the order number of the sensor(s) below between the parentheses,
    each line for new sensor)*/
    
    div.col-md-9.col-sm-8 > div:nth-child(3) > div:nth-child(1), /*fisrt sensor*/
    div.col-md-9.col-sm-8 > div:nth-child(3) > div:nth-child(6) /*sixth sensor*/
    {
    	display: none;
    }
    
    </style>`