Manuel83/craftbeerpi3 #29: Setup Note - Fermentation

helibrewer, 2017-06-19:
The chart tool associated with the fermenter displays Sensor (Sensor1). Therefore, if you are going to use multiple temperature sensors for your fermenter (wort, chamber, glycol), and you want to chart your fermenting wort, you must assign your wort probe to Sensor1.

carlallen, 2017-06-21:
My plugin uses 2 temp sensors. It's kind of annoying that I can't add them to the chart. It would be useful if plugins could modify the output of charts. I have an idea how that might be done, maybe I'll make a pull request and see if it gets accepted..

carlallen, 2017-07-26:
@helibrewer I made a pull request some time ago https://github.com/Manuel83/craftbeerpi3/pull/94 that adds the ability to modify what is displayed in the charts. My plugin now charts the setpoint, the beer temp, the chamber temp and the generated chamber setpoint.