Manuel83/craftbeerpi3 #37: Hardware Setup Not Recognized

beer4me, 2017-06-21:
I installed CBP3.0 on an image with CBP2.2.
I setup my hardware:
[image]
...and when I try to start a fermentation profile, I get this:
[image]

What did I get wrong?  I've rebooted several times.  Tried activating the Master Branch.  Nothing seems to work...

closed 2017-06-22 by beer4me

swimIan, 2017-06-22:
I think you have to add a fermentor with the switches and sensors associated with it.

beer4me, 2017-06-22:
Yep.  I watched the video closely and figured that out... goofy setup process... wanted an 'Actor' and a 'cooler'??  I'm running my fermentation chambers with a sensor for beer (in a thermowell) for control to heat or cool (season dependent) and another sensor to monitor a second beer or chamber temp - the 3.0 setup made no sense.  Also, setting up a fermentation session was much more cumbersome than 2.2... went back to 2.2...  Thanks for the reply. (I tried to delete this issue, but I'm new to github...)