Manuel83/craftbeerpi3 #40: No invert option

Mrpond1970, 2017-06-24:
Hi 

can't find a invert option on any screen when adding an Actor. Missing from CBPI 2.2

jalim, 2017-06-25:
Rather than having an invert checkbox for each actor the current convention is to choose an inverted actor.  For example  Relayboard is an inverted version of GPIOsimple and is available as default.  Alternatively there is a inverted PWM available as a plugin which you can install.  Hope that makes sense.

Mrpond1970, 2017-06-25:
Hi Jalim

I have the official board all setup and working so if i install the the inverted PWM i use that for my pumps and heater?

jalim, 2017-06-25:
It depends on how you have your relays wired but if you're finding that
they're on when they should be off then yes the inverted PWM plugin should
solve this.

Mrpond1970, 2017-06-25:
Thanks Jalim

I will give it a go and update later

Mrpond1970, 2017-06-26:
I tried the Inverted PWM tonight and still the same

jalim, 2017-06-27:
When you say still the same, can you detail the problem that you are experiencing?

Mrpond1970, 2017-06-27:
Hi Jalim

When the pump and heater buttons are in the off position the heater and pumps come on and when in the on position both are off. This was not a problem in 2.2 because i ticked the inverted boxs. I setup Inverted  PWM and still have the same results. I will be honest though I'm not sure how to set PWM up but i set the Cycles Per Second in - position and + position and it did not change anything.

Mrpond1970, 2017-06-27:
Inverted PWM - Just had a look at the Plugin details and its only for people running there heater relay in inverted mode. No Inverted mode on version 3 yet

Manuel83, 2017-06-27:
I will add an inverted PWM Plugin

fuzzewuzze, 2017-07-07:
+1 for this please.

carlallen, 2017-07-09:
You don't need an invert option. The RelayBoard actor type is identical to the GPIOSimple actor type except it is inverted.

If you need inverted GPIOPWM there is a plugin for that https://github.com/seesser/InvertPWM

fuzzewuzze, 2017-07-09:
Great ill try it thanks.

All the more reason for better documentation then i suppose.  
CBPI is amazing, but the lack of documentation about what things do needs some work :)

carlallen, 2017-07-09:
it's free and open source software, created by a community in their spare time. If you want better documentation, you are free to write it and submit it.

jtaylor54, 2017-07-10:
+1
Personally I would rather see an inverted checkbox (like hide) next to GPIO pin combo box so any add-on that used GPIO selector could just select inverted or not? That would save having to write an inverted version of some add-ons wouldn't it? To me having an extra add-on just to have inverted feature is confusing and may not be updated when the base add-on is. Relayboard is not intuitively the inverted version of GPIOSimple. At least have a naming convention like GPIOSimple (Inverted) instead and put for use with a Relayboard in the comments.

carlallen, 2017-07-15:
The problem with having a universal inverted checkbox is some actors don't have the option to be inverted. As an example: A wifi outlet is only going to have on and off.

jalim, 2017-07-15:
It might be sensible to have just a single GPIO and PWM actor with a drop down to select inverted?

carlallen, 2017-07-15:
in the case of the built in GPIO actors maybe. There isn't a built in inverted PWM actor... so it wouldn't apply there.

jalim, 2017-07-15:
I've not really looked at the PWM actor in too much detail but am I right in thinking you simply invert the duty cycle?  IE 100% duty cycle for an inverted PWM is actually 0% duty cycle?

jalim, 2017-07-15:
You're right, it needs a few changes.  I've noticed the term duty cycle being used in the PWM actor where it should really be titled frequency and may be confusing some people.

jtaylor54, 2017-07-16:
@carlallen 

> The problem with having a universal inverted checkbox is some actors don't have the option to be inverted. As an example: A wifi outlet is only going to have on and off.

But does that actor have a GPIO Pin Selection dropdown? I was thinking only have it when the GPIOPin Selection dropdown is shown.

germanherbik, 2018-07-27:
Hello, any solutions? i`m using GPIO Simple and y need invert de output signal in the GPIOs. I intalled the inverterPWM plugin but no is the solution.

germanherbik, 2018-07-28:
@HooTsBrooklyn i have this RelayBoard [https://www.switchelectronics.co.uk/5v-16-channel-relay-board-module](url) and the problem is when shutdown the raspberry, because when the raspberry is OFF all the relays are OFF, but when the raspberry y ON and the crafbeerpi is runing (when the actors OFF) all the relays are ON. 
If i will connect NC and COM when the raspberry is OFF y have all working! 

Sorry if i have bad english!

HooTsBrooklyn, 2018-07-28:
I have a similar board (https://www.amazon.com/gp/product/B01MT1MI8N).  Mine is switched with 5v and the GPIO pins only send 3.3v.  I had to use a ULN2803 (8-channel Darlington Array) IC chip to make it work.