Manuel83/craftbeerpi3 #42: Kettle frame for  boiling

AHm74, 2017-06-25:
Would it be possible to adapt a kettle frame for boilers? When boiling or taking the liquid to boil the temperature is not that interesting, the power is what I want to see and control.

Right now I'm using the target temperature to control the power to the boil (named temperature but controlling power) top right on the picture.

It would be nice if it was possible.

[image]

jalim, 2017-07-02:
Does the latest UI fix your request?  Power is now shown alongside the heater for the kettle.

[image]

Manuel83, 2017-07-02:
Should be available now

closed 2017-07-02 by Manuel83

AHm74, 2017-07-03:
Well, yes. This will for sure do it. For me the temperature is not really interesting at all while boiling. The boiling part is primarily a matter of setting the power to get to boil and after that to be able to control the boiling to a "good rolling boil".