Manuel83/craftbeerpi3 #45: Why not set actor.power to 0 in ActorAPI.switch_actor_off?

FredWanninger, 2017-06-26:
Great idea to enhance actors from simple on/off switches to abstract value-based switches! Finally I can use Craftbeerpi without hacking the code base :+1: 
But why don't you set actor power (the value that is visible in the UI) to 0 in ActorAPI.switch_actor_off?
For my solution (custom KettleController, custom Actor), i have to set the value to 0 via some workaround after calling heater_off to have the UI display the correct value.

jalim, 2017-06-26:
If you're using a custom kettle controller, self.actor_power(x) will change the power, you could just use this when ever the controller is told to stop?

jalim, 2017-06-26:
def stop(self):

        super(KettleController, self).stop()
        self.actor_power(0)
        self.heater_off()

FredWanninger, 2017-06-26:
Hm yes, actually that is what i do. This is a perfectly working solution. :)

But from my restricted perspective i would say calling two methods to do one single (atomic) thing is a workaround already.
I mean you have a method to switch on and pass a value in one step, but not to switch off and set the value to 0.

jalim, 2017-06-26:
Could be added as an option to send a power setting with heater_off() but I think it might be too much to presume that whenever someone turns off an actor that they want it set to 0, some may want it permanently set to 70% so it's ready to turn back on manually.

FredWanninger, 2017-06-26:
I think i understand the use case for setting the switch to off, while leaving the value constant now.
But it would be perfectly fine for me, if you could add the value as option to heater_off as in heater_on.