Manuel83/craftbeerpi3 #49: Server Freezing

swimIan, 2017-06-27:
Anyone else's craftbeerpi 2.2 or 3.0 server seem to crash ever so often. The menu items can be clicked on but I can't enable/disable any buttons on main page.  The Pi itself isn't frozen. On 2.2 it seemed to occur when switching kettles (mash to boil) that used the same gpio (I BIAB). On 3.0 the server freezes most often using Fahrenheit with PIDarduino and less often using Celsius. Any thoughts or suggestions? Does this happen to others?

closed 2017-07-02 by swimIan

peretti2004, 2017-07-09:
This situation is happening with me. The RPI is ok, but the CBPI crash, commands on screen does not work. It happens often.

peretti2004, 2017-07-10:
I had some issues in my setup hardware configuration, for example, I was using GPIOPWM instead GPIO Simple in a wrong way, and other little things. After read the wiki here and make the corrections, the cbpi are very stable.