Manuel83/craftbeerpi3 #52: Buzzer Not Working

Mrpond1970, 2017-06-30:
I can't get the buzzer to work with my Craftbeerpi board. set the buzzer up on GPIO 20 and nothing happens during the steps process. I have set the buzzer up as an Actor and when pressing the on screen button it works.

swimIan, 2017-07-02:
Same problem for me. Do the steps need to be rewritten to include a buzzer?

jalim, 2017-07-02:
Not all of the steps include notifications of step completion.  Some do and others don't.  Are you finding that it never beeps or just that it doesn't beep as the steps progress?

Mrpond1970, 2017-07-02:
I find that it never beeps

jalim, 2017-07-02:
Can you download your app.log and have a look for mentions of buzzer?  If it's having trouble there should be a log entry both showing which GPIO pin its trying to use and wether it is failing.

Mrpond1970, 2017-07-02:
Very Large App.log 66k

jalim, 2017-07-02:
If you do a search for any mentions of buzzer, should be easy to find a few.

Mrpond1970, 2017-07-02:
/home/pi/craftbeerpi3/modules/buzzer/__init__.py: 
2017-06-26 20:47:26,190 - INFO - INIT BUZZER NOW GPIO16
2017-06-26 20:47:26,191 - INFO - BUZZER SETUP OK
2017-06-26 20:47:26,191 - INFO - INIT OK
2017-06-26 20:47:26,192 - INFO - INITIALIZER - METHOD init PAHT 

Looks like GPIO16 being used but i have selected GPIO20 in the front end as my PI Board uses 20 for the buzzer

jalim, 2017-07-02:
Ah ok, have found the bug.  It appears that it's trying to read the buzzer setting before the setting has been loaded so it's just defaulting to 16.  I will have a play around and create a fix if I can.

Mrpond1970, 2017-07-02:
Great stuff thanks Jalim

jalim, 2017-07-02:
I have submitted a fix for Manuel to action when he gets a chance.

carlallen, 2017-07-07:
did the fix solve the problem?

fekeese, 2017-07-07:
Yesterday I installed a new version of 3 and the buzzer stopped, I was using the craftbeerpi 3 for two weeks and everything was perfect before that.

Wiring -  BUZZER on GPIO16 like default  :( it never beeps

jalim, 2017-07-07:
Check the parameters page has it set to 16.  There was a bug where it was defaulting to 16 no matter what was selected which has been fixed so now you'll need to select 16.

swimIan, 2017-07-08:
I brewer today with a fresh update. My buzzer on 20 did not beep at any point.

helibrewer, 2017-07-09:
I don't use a buzzer, but I do see an ERROR message for the buzzer in my log.
[app.txt](https://github.com/Manuel83/craftbeerpi3/files/1133835/app.txt)

fekeese, 2017-07-09:
Same as mine :(

swimIan, 2017-07-13:
Should line 16 be "gpio=self.gio" or should it be "gpio=int(self.gpio)" as it was in version 2.2?

PR #80 (2017-07-15): Use Integer for Buzzer GPIO
Closes #52
closed 2017-07-15 by Manuel83