Manuel83/craftbeerpi3 #58: Sensors 0 Degrees during Mash

Mrpond1970, 2017-07-03:
When using the heater and pump during the mash process in my Speidel Braumeister clone both my temperature sensors change to 0. This happens to the sensor in my kettle and the sensor in my fermentation chamber. It only happens when both operate at the same time. Boil on its own is ok and the temperature sensors are ok on there own.  I have pasted my log below showing my sensors working and the changing to 0. also an image of the main display. 

2017-07-03 20:59:29,831 - INFO - READ TEMP
2017-07-03 20:59:30,711 - INFO - READ TEMP
2017-07-03 20:59:34,342 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:34,349 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":77.06},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:34,352 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":77.06},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:34,353 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:34,355 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":20.69},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:34,359 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":20.69},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:34,476 - INFO - emitting event "SWITCH_ACTOR" to all [/brew]
2017-07-03 20:59:34,478 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":1,"type":"GPIOSimple"}]
2017-07-03 20:59:34,480 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":1,"type":"GPIOSimple"}]
2017-07-03 20:59:34,482 - INFO - emitting event "SWITCH_ACTOR" to all [/brew]
2017-07-03 20:59:34,484 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":0,"type":"GPIOSimple"}]
2017-07-03 20:59:34,485 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":0,"type":"GPIOSimple"}]
2017-07-03 20:59:34,711 - INFO - READ TEMP
2017-07-03 20:59:35,587 - INFO - READ TEMP
2017-07-03 20:59:39,360 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:39,367 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:39,370 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:39,371 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:39,373 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":20.63},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:39,375 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":20.63},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:39,486 - INFO - emitting event "SWITCH_ACTOR" to all [/brew]
2017-07-03 20:59:39,488 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":1,"type":"GPIOSimple"}]
2017-07-03 20:59:39,489 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":1,"type":"GPIOSimple"}]
2017-07-03 20:59:39,591 - INFO - READ TEMP
2017-07-03 20:59:42,471 - INFO - READ TEMP
2017-07-03 20:59:44,377 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:44,384 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:44,386 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:44,388 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:44,390 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:44,392 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:44,471 - INFO - READ TEMP
2017-07-03 20:59:44,490 - INFO - emitting event "SWITCH_ACTOR" to all [/brew]
2017-07-03 20:59:44,492 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":0,"type":"GPIOSimple"}]
2017-07-03 20:59:44,494 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":0,"type":"GPIOSimple"}]
2017-07-03 20:59:44,496 - INFO - emitting event "SWITCH_ACTOR" to all [/brew]
2017-07-03 20:59:44,497 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":1,"type":"GPIOSimple"}]
2017-07-03 20:59:44,499 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SWITCH_ACTOR",{"config":{"gpio":"19"},"hide":0,"id":1,"instance":1,"name":"Heater","power":100,"state":1,"type":"GPIOSimple"}]
2017-07-03 20:59:44,947 - INFO - df80db25fbc847bda859a7472d0bddc0: Received packet PING data None
2017-07-03 20:59:44,947 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet PONG data None
2017-07-03 20:59:47,351 - INFO - READ TEMP
2017-07-03 20:59:49,351 - INFO - READ TEMP
2017-07-03 20:59:49,393 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
2017-07-03 20:59:49,402 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:49,404 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-800000046a32"},"hide":0,"id":1,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Kettle Sensor","type":"ONE_WIRE_SENSOR"}]
2017-07-03 20:59:49,405 - INFO - emitting event "SENSOR_UPDATE" to all [/brew]
[... cut ...]
2017-07-03 21:00:49,583 - INFO - df80db25fbc847bda859a7472d0bddc0: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]
2017-07-03 21:00:49,585 - INFO - cdadef0995fe4e1aab192675d11c04c1: Sending packet MESSAGE data 2/brew,["SENSOR_UPDATE",{"config":{"offset":"0","sensor_name":"28-80000004723a"},"hide":0,"id":2,"instance":{"unit":"\u00b0C","value":0.0},"mode":"P","name":"Fermenter 1","type":"ONE_WIRE_SENSOR"}]

[image]

jalim, 2017-07-03:
Sounds like this could be an issue of EMF interference causing the signal from the temp probes to be lost.   If you're drawing enough current through your heater power cord and this runs near your sensor leads it could be enough to make them disappear.  Without knowing your hardware setup it is hard to guess what's wrong.

IndyJoeA, 2017-07-03:
Agreed.  I had a similar issue where the sensors would drop to 32 every time I turned off a pump, so I had to had an RF filter circuit to each sensor (resistor and capacitor).  That solved my problem.

Mrpond1970, 2017-07-03:
Indy I have seen this drop tonight as well.

Think you could be right Jalim I will have a look under the hood tomorrow and see if I can keep the sensor wire away from other wires.

Mrpond1970, 2017-07-04:
Do you think 12v for my pumps will also affect the sensors?

jalim, 2017-07-04:
My very very limited understanding of electro magnetic interference is that it is a function of the current and the voltage in the wire (plus a few other factors) so my guess would be that 12V pump should cause far less interference than the high voltage and current of your heating element.

Mrpond1970, 2017-07-04:
Thanks Jalim

I will have a look at it today

Mrpond1970, 2017-07-04:
I have changed my wiring tonight and used a shielded 3 core signal wire for my sensor from my kettle and it now works fine! 

Thanks for you help again Guys.

closed 2017-07-04 by Mrpond1970