Manuel83/craftbeerpi3 #60: Step-Timer stuck at 00:00:00

Hampfter, 2017-07-05:
The StepTimer doesn't show the correct time. It always displays 00:00:00. It is running for the correct amount of time, it may be just an GUI update issue. I could reproduce this on an Android Phone and a Laptop that was connected via Wifi in Chrome and Firefox.

umbertocristale, 2017-09-12:
I have the same problem and this seems to be a difference between raspberrypi systime and my laptop systime.

Philusha1983, 2017-09-12:
I made some tests with this issue here are some results:
1) If browser on RPI is running CBPi during brewing the timer will run on
all devices
2) If CBRi running from phone/laptop via IP:5000 the timer won't run.

I'm not a developer but I'm playing around with code and making bypass
solutions
and here is my possible solution (not tested yet)
when CBPi starts on boot add a vncserver command on terminal (will create
virtual monitor even if you not conecting monitor to pi)
then when raspianOS is started make a autostart command for chrome with
localhost:5000 url.

According to my tests, this might solve the the step-timer.

I think this issue is very old (had this on version 2.2) so it better be
solved correctly via code and not with bypass....

Philip   ;)

umbertocristale, 2017-09-13:
I put craftbeerpi3 running side-by-side on raspberrypi and on my laptop via IP: 5000 for some more tests and when the systime of raspberrypi and my laptop is the same, I noticed a second or two difference but when I changed the systime of my laptop for 3 hours later for test then step time was 00:00:00.
In my case I use the raspberrypi offline from the interner so raspberrypi systime is not synchronized.