Manuel83/craftbeerpi3 #63: Allow Actor or Sensor to access a StepProperty

IndyJoeA, 2017-07-05:
I noticed that there are StepProperty properties that give access to a list of all the actors or sensors from within a Step plugin, but when I try to use one of those within an Actor plugin, the dropdown does not show up.  Specifically I am trying to create an Actor that allows the user to select other Actors.  I think it would be useful for all the properties to be available to any plugin, no matter what type, if this is feasible.

```python
actor = StepProperty.Actor("Actor")
```

IndyJoeA, 2017-07-07:
@Manuel83 - I have identified the changes needed to make this enhancement work, however it will require a few small changes to the bundle.js.  Are you open to accepting pull requests to this file, or would you rather I post the code elsewhere so you can make the changes to the original source code?

Manuel83, 2017-07-07:
I will do changes in the bundle.js 

In general I like to rework the actor api. The big problem is to find a way that even the old version is working.

IndyJoeA, 2017-07-07:
My changes do not change the functionality of old versions or other plugins, rather they just make the dropdowns available outside of step plugins.  I will do a pull request so you can at least see what changes I made.

Manuel83, 2017-07-07:
I can do this change but than I will allow to access sensors, actors and kettles via drop down.
I will to it tomorrow if that would be ok?

IndyJoeA, 2017-07-07:
Yes that would be great.  Do you want to see my code then?

Manuel83, 2017-07-07:
i appreciate your offer but not necessary. I will change the core directly and add the missing line to the java script.

Manuel83, 2017-07-08:
Added for actors and sensor. Now you can use the d

[image]

[image]

closed 2017-07-08 by Manuel83