Manuel83/craftbeerpi3 #65: Fermentation control loop issues

torrdan, 2017-07-05:
I did use CBP3 for controlling a fermentation chamber (Fridge with one digital temperature sensor).
I tried to use both simple GPIO  and GPIO compressor configurations. 
With the automatic mode, both configurations gave the same issue: output should be zero (hysteresis logic and current temperature within range in which relay should be off) but there are some "pulses" to the fridge relay. Duration is about 1 s and interval is random, can be some seconds or minutes. I saw it because the relay light started to blink. This may result sin damaging the compressor. Would be great to fix it. Thanks.

carlallen, 2017-07-06:
the GPIO module has nothing preventing it from pulsing. If there's an issue with the GPIOCompressor plugin that should be reported there: https://github.com/carlallen/cbpi_GPIOCompressor. 

Do you have a pullup/pulldown resistor on your output pin, because input float can cause issues with gpio.

torrdan, 2017-07-06:
The same issue did happen also with simple gpio configuration. This is the reason why I posted it here. 
At the moment I do not have any pullup/pulldown resistors but I will add one to the fridge's pin. In any case it is a bit strange because with identical setup and CBPI2 I didn't have the same problem.

carlallen, 2017-07-06:
Do you have your output on the same pin as the buzzer? GPIO16, this could cause weirdness.

torrdan, 2017-07-06:
No, fridge output is on GPIO13 and I am not using a buzzer.  
Nothing is connected to GPIO16.

carlallen, 2017-07-07:
what do you have your offset set to?

torrdan, 2017-07-07:
I set the offset to zero

carlallen, 2017-07-07:
yeah, if you have the offsets at zero it's going to cause the output to strobe.

carlallen, 2017-07-07:
although GPIO compressor should prevent that strobing

torrdan, 2017-07-09:
Also with an offset for the temperature sensor the GPIO output is strobing ( with simple GPIO setting).
If I use GPIO compressor this results in the fridge not restarting after it turns off the first time

carlallen, 2017-07-09:
that's GPIOCompressor doing it's job. You specify a delay (in minutes) in the configuration, it won't allow  you to turn it back on until that many minutes have elapsed since the last time you turned it off.

jalim, 2017-07-09:
Just a thought, gpio compressor might not do the trick with a zero offset, it's going to turn on for an instant, then turn off because the temperature has bounced ever so slightly, then its inhibited for the off time.  I think the only answer here is to increase the offset so it has a chance, even half a degree might be enough?

carlallen, 2017-07-09:
yup @jalim 

I made a change a while ago that allows the offsets to be floats so you could make the ON offsets quite small (0.001) and it would fix your strobe issue, you can keep the OFF offsets at 0

torrdan, 2017-07-11:
There was a problem with the offset settings, maybe due to the decimal mark. Now with Cooler offset ON = 0.5, Cooler offset Off =1 and an offset of 0.1 for the sensor, the control loop is working fine. 
I did modify the the following line in the fermenter_hysteresis init.py file:
35: if temp <= target_temp - float(self.cooler_offset_max):
The sign minus comes from the fact that, to my understanding, the idea of an hysteresis control is to turn on and off the relay in order to maintain the temperature oscillating around the set value. With the plus sign the fridge is turned off "too early".

carlallen, 2017-07-11:
@torrdan why not just set your cooler_offset_max to a negative value?

torrdan, 2017-07-11:
sure, result it's the same but this is not in agreement with what is stated by the info on the gui for Cooler Offset OFF.
The info says that if offset_max = +1 the fridge will be switched off when the measured temperature is 1 deg below the target temp. This is not what is implemented on line 35 where fridge is turned off if measured_temp <= target_temp +1

carlallen, 2017-07-11:
sure, but all the descriptions are wrong. the second part of all of them is flipped between what they should say for heat/cool

torrdan, 2017-07-11:
I agree:-) 
@carlallen thanks for the prompt replays. 
I am going to close the issue since everything is working. Maybe it should be highlighted that the descriptions have be corrected.

closed 2017-07-11 by torrdan