Manuel83/craftbeerpi3 #68: Fermenter hysteresis

MikonosII, 2017-07-07:
The fermenter hysteresis requires values in the heater component even if no heater actor exists in the fermenter hardware setup. With out some values in for the heater on / off the auto won't work.

carlallen, 2017-07-07:
It works fine for me. Do you have values in all 4 offset fields? What does your log say?

helibrewer, 2017-07-09:
Auto mode works but you will see errors for the function heater_off(), I assume because no heater actor is present
[app.txt](https://github.com/Manuel83/craftbeerpi3/files/1133834/app.txt)

carlallen, 2017-07-09:
Yeah, it will show errors in the log, but shouldn't break anything. It's probably not a bad idea to verify that there is a heater configured before turning it on/off, but since that doesn't actually break anything it's pretty low priority in the fixes.

berazafi, 2017-07-17:
I had the same problem.  Without any values entered into the heating offset fields, the Auto function would not turn on the cooling

carlallen, 2017-07-17:
Yeah, even if you are not using heating, you still have to fill in all offset fields.

Fuempel, 2018-11-03:
This can be more easily detected when using PR #221