Manuel83/craftbeerpi3 #69: REVERSED GPIO Ports for Header and agitator

schuessm, 2017-07-07:
My configuration
Header GPIO 17
Agitator GPIO 22

I do this configuration but when i switch the Header on the agitator runs, perhaps the assignment in the kettle configuration is reversed

jalim, 2017-07-07:
Please check your hardware setup again and make for sure you've got the numbers the right way around.  I would suggest having a look at https://pinout.xyz if you want to check the raspberry pi connections.  Second thing would be to change your heater and agitator over to completely different GPIO pins and see if this works.

schuessm, 2017-07-07:
The Hardware is fine. I used the same configuration with CBPI2.2 and it works. And I just switched back to my version 22 and see that everything works, as before

carlallen, 2017-07-09:
Seems to be working fine for me. Re-check your actor configuration? Post pictures if you think it's still a problem

mrillies, 2017-09-22:
Is this still an issue? 

If so please post you craftbeerpi.db file here

schuessm, 2017-09-23:
No Problem when I use GPIOSystem-Plugin, I can check high or Lop and that do what the reverse function do

closed 2017-09-23 by schuessm

mrillies, 2017-09-23:
Just want to clarify, are you saying you needed to invert your output direction? How have you got it wired?

GPIOSimple vs RelayBoard are the same except RelayBoard is Inverted.