Manuel83/craftbeerpi3 #74: Using GPIOPWM for PID logic causes CBPi to exit

IndyJoeA, 2017-07-10:
While testing PIDArduino and PIDAutoTune, I noticed that if I have a heating element actor set to use GPIOPWM logic and then start the automatic mode for that kettle, a short time later CBPi will exit.  No exceptions are logged and nothing is printed to the terminal window, the program just exists.  GPIOSimple on the same heating element actor with PID works perfectly.  This is on a Raspberry Pi 3 running the latest Raspbian build.

carlallen, 2017-07-10:
Try running cbpi manually so you can see the errors without the log:
1) In the console stop the server run `/etc/init.d/craftbeerpiboot stop`
2) In the craftbeerpi3 directory run `python run.py`
3) You should now see an active log
4) Run your stuff with GPIOPWM

Do you get better error output now?

IndyJoeA, 2017-07-10:
That's actually how I first discovered the issue.  I was testing autotune by running `sudo python run.py`, and the python process would just exit with no output, and return me to the shell.  No exceptions, no tracebacks, etc.  I tried the same thing on a Raspberry Pi Model A (Gen 1 v1.2) and there I saw that it actually reboots the whole Pi.  So maybe it's something the GPIO library is doing that crashes a driver, or causes a power issue.  Not sure, but I have now read of other users encountering the same thing so I thought I'd post my findings.

carlallen, 2017-07-10:
are your relays connected directly to the pins on the pi or are you using a shield or transistor? it's possilble you could be over-drawing power on the pins.

carlallen, 2017-07-10:
oh man... looking at the code, I can already see a bunch of bugs...

swimIan, 2017-07-10:
I rewrote the PIDarduino code to not switch the heater on and off. I have it the PID equation setting the actor(heater) power instead and using GPIOPWM. I get no freezing or rebooting now. I modeled my code after the Hendi control logic btw. I use the same PID parameters as in 2.2 and I get very good temp stabilization.

carlallen, 2017-07-10:
I bet GPIO.PWM doesn't like start/stop being called when it's already started/stopped.

jalim, 2017-07-10:
I made a pull request to stop the pid logic turning off every cycle for an instant even when there was no need to.  I was seeing flickering on the ssr status led when I expected 100% power to ramp up.

IndyJoeA, 2017-07-11:
@jalim I see your changes to PIDArduino, that looks good to me.  I will test this out with my PIDAutoTune plugin and see if that solves the problem.  Of course it may be more of a work around if there is some underlying issues like Carl mentions.

adrianoftyriel, 2017-08-06:
Any update on this? I'm still experiencing the issue.

rechvagner, 2017-08-28:
@IndyJoeA I have the same problem.

I run a sample test with the Mod_PWM + PIDArduino plugins and it worked fine.