Manuel83/craftbeerpi3 #75: Where is the UI source code?

BoomerBrian, 2017-07-10:
Is there a reason only the bundled.js file is checked in? I would like to make some UI enhancements.

helibrewer, 2017-07-11:
I think the UI is completely generated by bundled.js

BoomerBrian, 2017-07-11:
I get that it is bundled. My question though is there a reason the unbundled source just isn't checked in? Does Manuel not want people working on the UI? That is what I am trying to find out. I am OK with whatever the answer is but I have asked this question both here and the Facebook group and nobody seems to know the answer.

bertah, 2017-09-19:
I would love to get Manuel's input on this too and would be very happy to see the full UI source checked in :)

SBassabe, 2018-01-18:
I do agree with all the comments. It is understandable if Manuel does not want to check-in the source code. It would just be nice to get a straight answer on the matter !!!

sretlawd, 2018-02-26:
No word on this?  I have some stuff I'd like to tweak too - is it just a matter of doing some code cleanup before exposing it to the world, or is the UI going to be a closed source project?

archonic, 2018-04-05:
The UI is the focus of most usability concerns. I don't see a good reason to not include the unbundled UI in the repo.

LucLN, 2018-05-17:
.. I see .. again no answer since Jul 2017 .. maybe [#186](https://github.com/Manuel83/craftbeerpi3/issues/186) ist the question to ask?