Manuel83/craftbeerpi3 #76: Bug when changing actor type

eskilmoen, 2017-07-10:
When changing actor type from PWMSimple to RelayBoard, UI throws error "Failed to set up actor.. Please check the configuration". 

Error is solved by re-opening the actor and re-setting the GPIO number. But the GPIO setting is lost so you have to check the board, or remember where it was configured.

Manuel83, 2017-07-11:
Are you running on a real raspberry pi?

niknokdereus, 2017-07-11:
Got the same here

eskilmoen, 2017-07-11:
Yep, latest version

carlallen, 2017-07-15:
@Manuel83 the UI doesn't clear the input field or reset the field so it's getting detached when it tries to submit. It only seems to be a problem if you change the type and the two types have the same fields with the same names