Manuel83/craftbeerpi3 #83: Buzzer Sounds

cocothenut, 2017-07-15:
Hi

where and how i can define various Buzzer-Sounds

for various Brew-Steps

in which file the Buzzer-Sounds are defined ?

what is the command to call the Buzzer-Sound ?

thanks for help

carlallen, 2017-07-15:
There's only one buzzer sound. You could make a plugin for additional buzzers sounds.

Currently in the code if you call `cbpi.beep()` it will beep the buzzer

swimIan, 2017-07-15:
I was thinking about modifying the buzzer to make multiple beeps rather than different sounds.

cocothenut, 2017-07-16:
Hi

my intention was not to add only some more beep´s

i want to add various Buzzer-Beep-Sounds , like in CBPi 2.2

for example :

def nextStepBeep():
    start_new_thread(playSound,(sound1,))

def timerBeep():
    start_new_thread(playSound,(sound2,))

def resetBeep():
    start_new_thread(playSound,(sound3,))

sound1 = ["H",0.1,"L",0.1,"H",0.1,"L",0.1,"H",0.1,"L"]
sound2 = ["H",1,"L"]
sound3 = ["H",0.5,"L",0.5,"H",0.5,"L"]

flo269, 2017-07-20:
How about placing a "Test buzzer" button somewhere in the ui? This way people without too much coding experience could easily test their setup before actually brewing... what do you guys think?