Manuel83/craftbeerpi3 #86: Gembird doesn't show up after install

flo269, 2017-07-16:
Dear all,

I've installed v3.0 and installed the GembirdUSBActor Plugin from the UI and also installed sispmctl from cli.

I've rebooted and still I can't choose Gembird from the drop down.
Any ideas what I could do?

[image]
[image]

carlallen, 2017-07-16:
Make sure you are running the latest version of cbpi3. In the System->System menu, click Activate next to master branch and then pull the update.

flo269, 2017-07-17:
I'm quite sure that I'm on current master now. Pulled, rebooted and still can't select Gembird.

[image]

flo269, 2017-07-17:
It looks like the repo https://github.com/Manuel83/cbpi-GembridUSB is empty. Or am I mistaking?

jalim, 2017-07-17:
You're right, Manuel may just be using it as a placeholder or it is a mistake.  Maybe raise an issue in that repository.

flo269, 2017-07-17:
ok, will do. thanks!

Redsak, 2017-12-27:
having the same problem here. Any Suggestions so far?