Manuel83/craftbeerpi3 #87: Autotune

rlcout, 2017-07-16:
Hello Guys,   change to version 3.0 and made a autotune. So they show me this numbers:
[image]
 
I think this's too high! any idea how to fix it?  thanks

carlallen, 2017-07-16:
This should probably go to the Autotune plugin: https://github.com/IndyJoeA/cbpi_PIDAutoTune

You can also ask questions in the facebook user group: https://www.facebook.com/groups/craftbeerpi

IndyJoeA, 2017-07-17:
I have seen another user with high P values.  I don't think it's necessarily a problem unless that value does not work well for you once you switch over to PIDArduino and input those values.  The PID algorithm is identical between those two plugins, so what works for one should work for the other.

rlcout, 2017-07-19:
well, thanks for answer.  Acctually I change all the values to that what I used in version 2.2 and works fine now.