Manuel83/craftbeerpi3 #97: No Beep beween Mash Steps

JamFfm, 2017-07-20:
There is a beep of the buzzer after reaching temp of Mash in step, but from Mash step to the next there is no beep. 
Got beep after updates or at the reboot. Love that.
It would be better with beeps beween Mashstaps so i do know that the next step starts without watching any Screen.

swimIan, 2017-07-20:
Does it beep when the timer elapses?

JamFfm, 2017-07-20:
no it does not beep when the timer elapses. This is the point.

swimIan, 2017-07-20:
I think a line needs to be added to the mash step to get the the additional beeps. Would you want the mash step to beep when it's a temperature (helpful for step mashing) and/or at the conclusion of the timer? Would you want the boilstep to beep in a similar manner (at temp, hop alerts, and at conclusion)?

swimIan, 2017-07-20:
I found the line to add and tested it. I'll create a pull request shortly. Stay tuned.

swimIan, 2017-07-21:
https://github.com/Manuel83/craftbeerpi3/pull/98

JamFfm, 2017-07-21:
The beep shouls appear at the end of the mashstep this would be the same as end of timer.

swimIan, 2017-07-21:
The pull request does add the beep at the end of the mash step

JamFfm, 2017-07-22:
perfect, sorry did not know that notification is combined with beep. 
Thanks for your work in this comunity

JamFfm, 2017-07-24:
Works for me. Leave it some days open in case someone has got an issue.

JamFfm, 2017-07-28:
Works so far. Does not beep when timer starts. But his is ok. Just want to mention in case someone says it is a failure.

closed 2017-07-28 by JamFfm