Manuel83/craftbeerpi3 #100: Empty webpage after adding Kettel

Nauticno, 2017-07-25:
I have installed 3.0 to a new CFcard.
Started with temperature sensors, all OK
Then actors, all OK
When adding kettel, the webpage is emty.

I have to clear the database, stop cbp and then start cbp for starting over.

What are I doing wron?
[image]

Nauticno, 2017-07-25:
I dowloaded a zip file from code section and overwrite the complete catalog. Now its workin, but git pull does not work.,

carlallen, 2017-07-26:
This might be a better discussion for the facebook users group. I'm not sure I would consider your specific case to be a code issue, especially as you got it to work with the zipped code.