Manuel83/craftbeerpi3 #104: Logic for fermentation

tomcasp, 2017-08-01:
When I run the system for fermentation controll it seems that the logic is not running correct.
The inpus is as following:
Cooler offset Max: 0.5
cooler offset mini: 0.51
heater offset max: 0.5
Heater offset min: 0.5.

When I sat the target temp for 18 deg Celsius the system didn't start the heater when the temp droped below 17 deg or started the cooler when the temp was 21 deg celsius.
When I run the heater and/or cooler manually everything is working, so I'm guessing that it could be an issue regarding the logic?

Best Regard Tom Caspersen
[image]

lalo-uy, 2017-08-01:
If you run the steps the fermentor do not start the AUTO. 
For me this is the bug or I do not understand the logic behind this.
Also if the fermentation is running and the system reboots the AUTO do not start.

lggrizendi, 2017-08-14:
I think in the system should be a option to put the auto as default in fermentation, i don't know is is possible. In my city there as lot power lost and my fermentation system don't work if not in this way.

lalo-uy, 2017-08-14:
It's just to set Auto on if there is a Fermentation-Step running.
I have do some patches, but not ok yet.

lggrizendi, 2017-08-15:
But i think if i put some target temperature in the fermentation the system  should be on auto or the target temperature chosen don't work.  this should work if the fermenation-step is no running. 
Sorry for bad english

tomcasp, 2017-08-16:
I'm now running the other version of this program, 2,2. The main reason for this is that the 3,0 won't start when I press "play". 
It seems like the program won't start the logic for the temperature control or follow the temperature schedule. 
I do hope that this is an issue that will be fixed, since it seems like the 3,0 is a much better version of cbp.

helibrewer, 2017-08-30:
Your screen shot only shows heating and cooling buttons, I do not see the auto button

peterblaha1, 2017-09-11:
It does not work on the automatic when I press play.  I expect a correction soon.

tomcasp, 2017-09-12:
@helibrewer ; I'm sorry for that the picture is not showing all the buttons, but the button is not missing. The logic is missing. As @peterblaha1 is saying, the logic does not work when 'Play' is pressed. 
I'm still running 2.2 and do not know if this issue is solved, but it seems to be unsolved.

lalo-uy, 2017-09-12:
Also the logic don't star if a reboot is done and the Fermentation step is
running.

mrillies, 2017-09-22:
There seems to be a couple of issues being discussed here. 
At this point CBPi does not support autostart at power-on.
This would be nice, agreed but is not the topic of this issue. If there is not a matching issue for this please start another one.
If you have many power issues i recommend you add battery backup to the pi. Using a USB battery bank is probably the easiest You will need one that allows charge and discharge at the same time.

As far as Logic issues:

I have experienced the press play not working. The traceback and lof show:
```
"/home/pi/craftbeerpi3/modules/fermenter/__init__.py", line 162, in start_fermentation
    if s.state == 'A':
AttributeError: 'dict' object has no attribute 'state'
```
This problem went away when i restarted, not sure what has caused this but will keep looking.

I see in the screenshot above that there is no "CAR" symbol which is the auto button. 
When you press play on the recipe list it will change the target temperature of the kettle, but unless the car is active the elements will not follow the target temp (this is the same as the brewing tab). Try refreshing the page. 

Make sure there is a valid number in each of the heating and cooling boxes for hysterisis. The on and off should be different numbers. The OFF is difference from set to switch off and On is when it switches on again. 
So OFF 1 and On 2 for both and set point of 20 means 
heating turns on when below 18 and turns off at 19, 
cooling turns on above 22 and off under 21

Can you please post a copy of your craftbeerpi.db and logs/app.log from the not working system.

mrillies, 2017-09-22:
I have narrowed down the bug.

Reproducible:
Create Fermentor
Add Steps
Steps are of type FermenterStep
Go to Hardware, update Ferementer
Steps are now dictionaries
Add a new step
New step is of type FerementerStep.
Restart CBPi, all steps are of type Fermenter Step again.

Summary: The "Update" Fermenter button mangles any steps on that fermenter created before clicking update

mrillies, 2017-09-22:
Possible fix proposed. It fixes the issue but may not be the best way. @Manuel83 can you please have a quick look. 
Basically the PUT method on fermenter update overwrites the fermenter step objects with the json.dumps version. I added a call to reset the fermenter which rebuilds the step attribute.

lalo-uy, 2017-09-22:
>
> When you press play on the recipe list it will change the target
> temperature of the kettle, but unless the car is active the elements will
> not follow the target temp (this is the same as the brewing tab).

Whats the point in starting a recipe and not going to Auto?

I belive that the Auto should be set at Play press.

mrillies, 2017-09-23:
There are two systems at work, a fermenter following a set point, and a recipe step that changes that set point.

Think of the Car as an 'Auto/Manual' selector.
You can turn off the auto and operate the actors manually without interrupting the recipe,  same as the brew stand. This type of control is very standard in process control solutions. 

There are 3 possibilities:
1) The Auto button turns on when you press play:
  \+ dont have to press one more button
  \- auto may be turned off for a good reason 
2) The auto button is on from power up 
 - Same +/- but means you can turn auto off once and it wont auto switch back on
3) The user switches the auto themselves
 \+ have control over whether a chamber is following its target temp or not 
 \- you have to press one extra button to enable fermentation

At this point, option 3 is the simplest from both a UI and backend view without needing to add a bunch of configuration properties.

viyh, 2020-01-05:
I just had this issue and found the thread. I expected the fermenters set to auto to start up again when the software was restarted or after a power failure. 

Just in case anyone wants fermenters to auto-start when CBPi comes back up, I wrote a quick plugin that provides that behavior here: 

[https://github.com/viyh/cbpi-autostart-fermenter](https://github.com/viyh/cbpi-autostart-fermenter)

It's not 100% on topic with this thread's original issue, but hopefully it will help someone else that stumbles upon this thread.

lalo-uy, 2020-01-05:
Nice apprach. 
I also have a plugin  ( https://github.com/lalo-uy/SupervisedFermentor ) that autostarts the fermenter if it have an Active step running.
No need to mess in hardware setup.