Manuel83/craftbeerpi3 #106: Cannot Access Custom Plugin

BrauereiAlt, 2017-08-10:
May be a permissions issue similar to #22.  Tried changing ownership of folder (directory) and files from user to root without effect.  Plugin is a modified copy of PIDBoil named PIDBoilwithRamp.  Directory by that name includes both __init__.py and README.md files saved by nano, as well as auto-generated __init__.pyc file.

carlallen, 2017-08-16:
what about it can you not access? Non-approved plugins won't show up in the plugins area in the UI, but their functionality should still work.

BrauereiAlt, 2017-08-17:
Understand that it won't show on the Plugin page unless published through Martin, but couldn't see it as an option in the pick list for a kettle logic on the hardware setup screen.

BrauereiAlt, 2017-08-17:
RESOLVED - user error.  I didn't rename the class when modifying an existing file.  You must use a unique class name.  The logic pick list from the Hardware Setup screen recognizes and populates the class name and not the directory name.  Maybe this could be better noted or emphasized in the Wiki/documentation.

closed 2017-08-17 by BrauereiAlt