Manuel83/craftbeerpi3 #109: Set Temp Resolution

dave-robinson-022, 2017-08-17:
Would it be possible to set temp at 0.1 degree resolution as I need to control a still valve and want to turn it on at 78.4 and off at 78.6 C

swimIan, 2017-08-18:
Yes. Just make sure to put a zero before the decimal.

dave-robinson-022, 2017-08-18:
Sorry I wasn't very clear; trying to set temp on kettle and it only has slider? maybe another slider underneath that allows 0.1 degree resolution would be good? I can't find the code for the set target temp slider?  My control logic is just a simple modification of hysteresis.

carlallen, 2017-08-23:
If you are using the onewire temp sensors, they don't have very good precision and tend to bounce on temps a lot. That's a pretty tight range for the sensors.

mrillies, 2017-09-22:
Id prefer the option to set via text box or with +/-.

As a short term fix you can use something like Postman in Chrome to send a POST request such as:
`http://ip.of.cbpi:5000/api/kettle/1/targettemp/23.1`
To set kettle 1 to 23.1 degrees. This needs to be a POST request not GET.