Manuel83/craftbeerpi3 #116: Update Folder .swl files

swimIan, 2017-08-29:
I was testing some small changes to the 3.0 base code in order to add greater automation through default actors (pump, etc) being inserted onto my steps. I added a new .sql file to the "update" folder but my code changes did not result in what I had hoped for. 

The insert or ignore for the default mash and boil kettle give a drop down of available kettles. I tried the same but used 'actor' where it had said 'kettle' but there is no drop down list available.

Two Things:
1) I'd like to get my improvements fixed but I'm not sure how to proceed help?
2) I'd like to undue these changes. I deleted the new .sql file but that did not remove the new parameters as I had expected. How would I do this?

carlallen, 2017-08-30:
the sql files are only intended to run once, if you make a mistake in your sql file, you are going to have to correct it in your database. The easiest way is to delete your existing database and it will automatically get re-created next time the application starts. If you have your code viewable somewhere, it would be easier to provide better feedback.

swimIan, 2017-08-30:
I tried some DELETE and UPDATE commands to remove my errored code but I was unsuccessful.  I may reset the database entirely as you suggested.  When added to default actors (pumps in my case), here is what I used.

note: This was added in schema_info.sql

INSERT OR IGNORE INTO config VALUES ('step_mash_agitator', 1, 'actor', 'Default Mash Agitator', NULL);
INSERT OR IGNORE INTO config VALUES ('step_boil_agitator', 1, 'actor', 'Default Boil Agitator ', NULL);

note: I did remove the extra space on the last line.  Also I did notice in the schema.sql file that the "actor" table isn't created until after the "config" table is created.  I wondered if this had something to do with the drop down list not performing as expected.  Substituting 'kettle' for 'actor' seems to work in schema_info.sql.  By extension, I thought that maybe this is what is affecting the ability to hide sensors on the dashboard.

carlallen, 2017-08-31:
It's possible that the actor type isn't implemented for the config section

swimIan, 2017-09-09:
Hi Carl,
Did you mean the config sql database, the cbpi cache, or elsewhere?