Manuel83/craftbeerpi3 #117: Error Loading LCDDisplay after latest CBPI update

JamFfm, 2017-08-31:
I get a Error like: Failed to Load plugin LCDDisplay. UNIQUE constraint failed: config.name
This is also shown in the applog. LCDDisplay is only a backgroundtask and not initialised at the beginning.

carlallen, 2017-09-05:
I reported as issue for the plugin: https://github.com/breiti78/craftbeerpiLCD/issues/12

breiti78, 2017-09-05:
I tried to fix it but I don't get it. I've added the an initalization at the end of the __init__.py file. Like this:

@cbpi.initalizer(order=1)
def init_lcddisplay(self):
    app.config['LCD_Adress'] = set_lcd_adress()
    app.config['LCD_Refresh'] = set_parameter_refresh()
    app.config['LCD_Multidisplay'] = set_parameter_multidisplay()
    app.config['LCD_Singledisplay'] = set_parameter_id1()

You can see it in the branch I have created
https://github.com/breiti78/craftbeerpiLCD/blob/Fix-Initialization-Issue/__init__.py#L217

carlallen, 2017-09-05:
yeah, the code in lines 27-43 is all going to need to get moved to the initializer

carlallen, 2017-09-05:
and the code on 53-54, 64-65, 75-76, and 110

breiti78, 2017-09-05:
Thanks. I'll do it tomorrow.

JamFfm, 2017-09-06:
Hi the one thing which worked for me is putting all the code in the initializer. 

@cbpi.initalizer(order=2100)
def init(cbpi):
    
    def set_lcd_address():
      adr = cbpi.get_config_parameter('LCD_Address', None)
      if adr is None:
          cbpi.add_config_parameter('LCD_Address', '0x27', 'string', 'Address of the LCD, CBPi reboot required')
          adr = cbpi.get_config_parameter('LCD_Address', None)
      if len(adr) != 4:
          cbpi.notify('LCD Address is not 4 digits', 'Change LCD Address in parameters', type = 'warning', timeout=None)
      return adr
    LCDaddress = int(set_lcd_address(),16)
etc...........
    cbpi.app.logger.info('LCDDisplay  - LCD_Address %s %s' % (set_lcd_address(),LCDaddress))

JamFfm, 2017-09-06:
The bad thing is that on restart during masstep the LCD sometimes is not showing anything.

JamFfm, 2017-09-06:
Tryed carls suggestion. There is the LCD Object missing when trying to write the lines.
Do i need to do things like init(cbpi).lcd.write_string(line1)?  
How do i get the values of the parameter in the Def Init indent in the "other " block? 
As you can see I am a rookie ;-)

breiti78, 2017-09-06:
Plugin has been updated. Thank you JamFfm.

JamFfm, 2017-09-07:
Thanks breiti78. This is teamwork which I like. 
I noticed a problem with the error handling and emitting notifys.
Working on the Problem and I m close to the solution

JamFfm, 2017-09-08:
I posted a better more stable version on breiti78 git hub. Still cant handle git sorrry breiti.

carlallen, 2017-09-12:
This is not an error in cbpi and I believe it is now fixed in the plugin. This should probably be closed.

JamFfm, 2017-09-12:
Yep is fixed but for new build in LCD there is a new version needed which is to be submitted via the plugin. Otherwise you cant keyin the new I2C adress, because application stops working and you cant change on prameters-page

closed 2017-09-12 by JamFfm