Manuel83/craftbeerpi3 #121: Request pump plugin

Blom1974, 2017-09-09:
Hoping that some one is able to do a simple plugin fur pump cycles like Brewmeister.

I would like to have a button or step where i turn on the pump.
The pump should then run for 60sec stop for 30sec and so on. 
Would be nice if the values are configurable of course.

Hoping for someone skilled to build it.

bertah, 2017-09-23:
Assuming your pump is controlled by GPIO, I believe this could be achieved by using the GPIOPWM actor. When creating the Actor, set the "Duty cycle" value (label is misleading, it should read "Frequency" - will be fixed soon, see Issue #21) to 0.017, which translates to roughly 60sec (1/60Hz). Then, when running the actor, set the power to 50% to get 30sec on / 30 sec off.

swimIan, 2017-09-23:
Have you tested the 0.017 frequency? Are we sure that's how the pwm portion rpi.gpio works that way? I only ask because I'd honestly like to know.

bertah, 2017-09-23:
I'm currently running my heater (stock GPIOPWM) with a 0.1 frequency ("duty cycle") which, at 50% power, gives me 5sec on/5 sec off (0.1Hz = 10sec). So I can confirm that RPi.GPIO can handle these ranges of values.

swimIan, 2017-09-23:
Nice thanks. 1 HZ is typically defined as 1 cycle per second. Is Frequency in rpi.gpio defined the same? Typical AC current is 50/60 hz.

bertah, 2017-09-23:
Yep, the frequency parameter in RPi.GPIO is in Hz, I found some examples here:
https://sourceforge.net/p/raspberry-gpio-python/wiki/PWM/

closed 2018-03-17 by Blom1974