Manuel83/craftbeerpi3 #128: running in cloud, using remote sensors, actuators

universam1, 2017-09-22:
Hi @Manuel83 

Love your new concept with extensible plugins, thinking about following setup:

- CBP runs in the cloud on any hoster
- actuators and sensors remote by plugins

Example: using http://www.exp-tech.de/itead-studio-sonoff-th16 via MQTT which becomes a remote sensor and actuator. 

Benefit:
- full solution for 12EUR
- scales extremly

downside:
- requires internet

Makes sense?

Greetings,

the iSpindel Sam

mrillies, 2017-09-22:
More possible downsides:

- Requires smarter actors or high network usage to Pulse width
- All sensors and actors will need to be smart, increased cost as scalled
- Need to pay for hosting.
- Timing issues for PID, PWM and steps
- Requires a tunnel/secure authentication or port forwarding.

That said, you should be able to do this now anyway, just install it on hosting with python support or on a PC running linux.

Manuel83, 2017-09-22:
A first MQTT Plugin is avaibale. https://github.com/Manuel83/cbpi-mqtt
It's up to you where your hosting the MQTT Message broker. It can run on your local Pi or in the Internet. In general you can run CBPi on any Windows, Linux or Mac OS. 
From there you can connect all sensors and via MQTT. That means you don't need a PI to run CBPI.

With the next version the MQTT Support will be extend.

universam1, 2017-09-22:
Thats amazing, especially the MQTT support opens up great opportunities!

Regardless of hosting or on-demand, having the possibility to connect all IO wireless is a game changer! The Sonoff seems to be like the choice for a cheap solution, but other options are there as well. Will give it a try.

universam1, 2017-10-26:
To give you some feedback @Manuel83 @mrillies - the setup works successfully for me in meantime!

So I used a Sonoff Basic and soldered an DS18b20, flashed with https://github.com/arendst/Sonoff-Tasmota so it became a MQTT smart Sensor and Actor. With a Sonoff TH x no soldering would be required.

Connected via MQTT and I am running it as a fermentation Sensor & Actuator currently, totally happy with this setup. For brewing should be working as well. All over ordered from China, I payed less than 4EUR per unit, I think you can't get cheaper.
Still, currently a patch is needed https://github.com/Manuel83/cbpi-mqtt/issues/3

Philusha1983, 2017-12-03:
@universam1 Can you share the workflow to build the wireless sensor? 
I have several fermentation vessels and Ispendel fits only one type, so I need to build more wireless temp sensors.
Thanks