Manuel83/craftbeerpi3 #131: Duplicate code in __init__.py

zanberdo, 2017-09-24:
brewapp/__init__.py has code duplication at lines 23..26

mrillies, 2017-09-25:
craftbeerpi3 does not have "brewapp", 

are you talking about V2?
I looked ar brewapp/__init__.py and 23...26 has no duplication.

zanberdo, 2017-09-25:
Manuel,

Slip of the finger. Yes, the fix I'm proposing belongs in
craftbeerpi/brewapp/__init__.py.

Specifically looking at lines 12 through 27 it seems pretty clear there was
a cut-and-paste error that allowed lines 14-17 to be repeated at lines
23-26.  Unless you intended to import wraps from functools and define app
as an instance of Flask twice, but i don't think you had.

I noticed that there wouldn't really be any indication of a failure as the
initial logging will log lines 21 and 22 before app is reinitialized and
log line 27 would be written, not giving any indication in the log files of
an error.

I was also looking for unit and functional tests, but didn't find any.
Question: is development of craftbeerpi still underway, or have all
development efforts been shifted to craftbeerpi3?  My intent is to install
cbp3 onto a RPi3 along with a single-wire thermocoupler and a pair of SSRs
to control our fermentation refrigerator.  Adding a 7" touch screen to the
door to permit interaction directly with the module, as all we access over
the web.

Thank you for your project, looks like it could be a very useful solution.

Cheers,
Mark

mrillies, 2017-09-25:
Hi Mark,

Jake here, you're right, looks like a copy paste slip. It won't affect functionality at all and I don't belive any dev is happening on 2.2 concidering theist commit date and the open pull requests. 

Manuel is one man team with a few of us self appointed helpers floating around. My recommendation is ignore 2.2 completely and move to 3, it's a complete rewrite. Manuel may have a private test suite but as it's private project that's gone viral it dosent have much official documentation.

zanberdo, 2017-09-25:
Jake,

Thanks for the response and feedback.  Yes, our intention is to deploy v3
when we complete the build.  I was just familiarizing myself with the
source when I found that minor issue (yes, I recognize it shouldn't have
any significant effect so long as reinitializing Flask doesn't do something
unexpected, which in this case I can't see it would).  I was a little
concerned that this had slipped through the cracks, but understand how it
could and why it wasn't caught.

My hope, once I've had a chance to read through v3 code and get my own
project built, is to contribute to the project where I can.  I'm a seasoned
software engineer and have been working with Python for the last couple of
years to write all my functional tests for my Java based SDN project at
work.  I was really hoping to find out how to become a contributor.  If you
have any suggestions to that end, please let me know.  In the mean time I
will focus my attention on v3 (although honestly it couldn't hurt to fix
the 2.2 issue :) ).

Cheers,
Mark

mrillies, 2017-09-25:
Hi Mark, and welcome :) 

Manuel is currently the only one with permissions on the repos but i try to submit constructive pull requests to keep any minor issues in check :) 

Development has been moving pretty fast and focused on adding new features, your expertise could be useful!

The CraftbBeer Pi user group on Facebook is a good place to help out as well.

I noticed there's a pull request there that has a few fixes to that file, not all of them required. If you want to do the edit and make the pull request go ahead but nothing's been merged over there in 6 months.