Manuel83/craftbeerpi3 #135: Problem with iSpindel Plugin - Display Values

hippohop, 2017-10-02:
I have a clean new installation of Raspbian Stretch and Craftbeerpi3.
After successfully installing the iSpindel plug-in and subsequent reset, I can configure Craftbeerpi3 with a new sensor for temperature, tilt, battery, etc., but no values are displayed.
iSpindel is, of course, configured on Craftbeerpi, a corresponding IP address including the default port 5000 and the corresponding wi-fi network.
There are also no values in the log file.
I do not know what's wrong or what I'm doing wrong.
How can I verify the correct communication iSpindel with Craftbeerpi3?

i-Spindel is functional because it currently works via UBIDOTS.
i-Spindel currently has the latest firmware 5.5.0.

thanks for any ideas

isarvalley, 2017-10-03:
I was running into the same problems. But I guess the cause is the new iSpindel Software. After a downgrade to 5.1.2 it worked again.

mrillies, 2017-10-05:
When you configure the iSpindel, do not put the port on the ip address. when you select Craftbeerpi as the server tyoe it adds the 5000 itself. 

the biggest issue with thtis is you must run CBPi on 5000 for ispindel to work

larsan, 2018-06-09:
We have the same issue, while working on the same network with the same iSpindel setup it works with Craftbeerpi 2.2 but not with CBPi3. iSpindel Firmware is currently latest 5.8.6.

I am not well versed in python debugging however netstat outputs that craftbeerpi3 listens on port 5000 and starting the run.py manually with strace outputs every time the iSpindel sends data:

```
[…]
epoll_wait(6, [], 1023, 99)             = 0
epoll_wait(6, [], 1023, 0)              = 0
epoll_wait(6, [], 1023, 99)             = 0
epoll_wait(6, [], 1023, 0)              = 0
epoll_wait(6, [{EPOLLIN, {u32=3, u64=1890971021213699}}], 1023, 99) = 1
epoll_ctl(6, EPOLL_CTL_DEL, 3, 0x7ee70610) = 0
accept(3, {sa_family=AF_INET, sin_port=htons(9957), sin_addr=inet_addr("192.168.178.181")}, [16]) = 7
fcntl64(7, F_GETFL)                     = 0x2 (flags O_RDWR)
fcntl64(7, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
fcntl64(7, F_GETFL)                     = 0x802 (flags O_RDWR|O_NONBLOCK)
fcntl64(7, F_SETFL, O_RDWR|O_NONBLOCK)  = 0
write(2, "(759) accepted ('192.168.178.181"..., 41(759) accepted ('192.168.178.181', 9957)
) = 41
accept(3, 0x7ee7093c, [16])             = -1 EAGAIN (Resource temporarily unavailable)
epoll_ctl(6, EPOLL_CTL_ADD, 3, {EPOLLIN|EPOLLPRI|EPOLLERR|EPOLLHUP, {u32=3, u64=1890971021213699}}) = 0
setsockopt(7, SOL_TCP, TCP_QUICKACK, [1], 4) = 0
recv(7, "POST /api/hydrometer/v1/data HTT"..., 8192, 0) = 354
getsockname(7, {sa_family=AF_INET, sin_port=htons(5000), sin_addr=inet_addr("192.168.178.42")}, [16]) = 0
send(7, "HTTP/1.1 204 NO CONTENT\r\nContent"..., 142, 0) = 142
shutdown(7, SHUT_RDWR)                  = 0
close(7)                                = 0
epoll_wait(6, [], 1023, 27)             = 0
epoll_wait(6, [], 1023, 99)             = 0
epoll_wait(6, [], 1023, 0)              = 0
epoll_wait(6, [], 1023, 99)             = 0
[…]
```
…181 being the iSpindel, …42 being the Pi. The data obviously reaches the craftbeerpi software, however I have no Idea how to debug it any further.

The packets that are being received seem to contain reasonable data though:
```
0000   b8 27 eb 41 1b fc 60 01 94 22 a8 d3 08 00 45 00   ¸'ëA.ü`.."¨Ó..E.
0010   00 b1 00 06 00 00 80 06 54 10 c0 a8 b2 b5 c0 a8   .±......T.À¨²µÀ¨
0020   b2 2a 64 50 13 88 00 00 1a 4e b1 c1 da e6 50 18   ²*dP.....N±ÁÚæP.
0030   16 d0 d7 4f 00 00 7b 22 6e 61 6d 65 22 3a 22 69   .Ð×O..{"name":"i
0040   53 70 69 6e 64 65 6c 5f 53 30 22 2c 22 49 44 22   Spindel_S0","ID"
0050   3a 32 32 37 31 34 34 33 2c 22 61 6e 67 6c 65 22   :2271443,"angle"
0060   3a 36 35 2e 39 30 39 38 38 2c 22 74 65 6d 70 65   :65.90988,"tempe
0070   72 61 74 75 72 65 22 3a 32 39 2e 38 31 32 35 2c   rature":29.8125,
0080   22 62 61 74 74 65 72 79 22 3a 34 2e 30 31 38 34   "battery":4.0184
0090   39 39 2c 22 67 72 61 76 69 74 79 22 3a 31 32 2e   99,"gravity":12.
00a0   37 38 35 31 35 2c 22 69 6e 74 65 72 76 61 6c 22   78515,"interval"
00b0   3a 36 30 2c 22 52 53 53 49 22 3a 2d 35 36 7d      :60,"RSSI":-56}
```
I'd like to assist further in debugging this, though I currently do not know how.