Manuel83/craftbeerpi3 #136: Fermenter do not run after reset

nambuco, 2017-10-02:
Fermentation is a long process and I expect Craftbeerpi can recover well from a power failure, but this is not the case. 

After reset, you have to manually restart the fermenter "auto" state, otherwise there will be now temperature control anymore. At 2.2 version I also had same problem and made I change in the code. I believe many have same problem and an official solution would be great.

lalo-uy, 2017-10-03:
Paulo, eu acho igual que Vc, por encuanto a solucaon oficial e "procure una
UPS"  :-(

rechvagner, 2017-10-09:
I haver te Sam e problemas.  A solution to this is interesting.

HooTsBrooklyn, 2017-12-09:
I'd be interested in this as well.  Using a UPS is not a solution as I don't want to run my keezer / fermentation chamber on a UPS for ANY amount of time.  Ever.

vaughankg, 2017-12-11:
@nambuco what change did you make?

I'm using craftbeerpi to run a keezer along with a fermentation chamber. I have flowmeters on the taps which also reset after a restart.

nambuco, 2017-12-13:
@vaughankg , I made the change in 2.2 version, but I lost my 2.2 installation and do not have a backup...

Ins0x, 2018-11-06:
Hey guys, no official solution yet? Any unofficial solution??

Ins0x, 2018-11-06:
Hey guys, no official solution for 3.0 yet? How about unofficial solutions??

lalo-uy, 2018-11-07:
You can try this plugin

 https://github.com/lalo-uy/SupervisedFermentor