Manuel83/craftbeerpi3 #149: ISpindel Plugin - Hide Sensor on Dashboard not works properly

BREWERNIE, 2017-10-19:
When a ISpindel is created and die checkbox "Hide on Dashboard" is set, the sensor will not displayed on the Dashboard.
Everything seems fine. No ISpindel sensor is shown on the Dashboard.

After restarting the system, the created sensor is now shown on Dashboard although the Checkbox "Hide on Dashboard" is set.
[image]
[image]

Philusha1983, 2017-10-19:
I have the same problem.
I cant hide sensors on dashboard.
I just hide them with custom CSS that I wrote in the index.html file

ᐧ

IndyJoeA, 2017-10-24:
This is already a reported issue that affects any sensor, not just iSpindel.  See issue #27

Philusha1983, 2018-11-07:
I made a CSS hack to hide the required sensors:
please take a look at the video:
https://youtu.be/0_a43LK0PVg 

the code:
`    <style>

    /*hide on dashboard hack
    write the order number of the sensor(s) below between the parentheses,
    each line for new sensor)*/
    
    div.col-md-9.col-sm-8 > div:nth-child(3) > div:nth-child(1), /*fisrt sensor*/
    div.col-md-9.col-sm-8 > div:nth-child(3) > div:nth-child(6) /*sixth sensor*/
    {
    	display: none;
    }
    
    </style>`