Manuel83/craftbeerpi3 #152: 30amp and 2 heaters

chrisbrucewilliams, 2017-11-01:
I'm planning a 10gal system using Blichmann pots and their heaters and my plan is to run this on 30amp breaker.  30amps will not support running both heaters at once yet the UI would allow me to turn them both on.  How do you prevent this from happening?

jangevaare, 2017-11-01:
While it wouldn't be too difficult to write a plugin to do this, my suggestion would be to look for a hardware solution to this (i.e. an element selector switch)

lalo-uy, 2017-11-01:
A selector switch is much more secure than a soft block

chrisbrucewilliams, 2017-11-01:
Today I looked at the "actor groups" that is being developed.  IMO it would make a ton of sense to be able to add logic statements to this.  Even simple things like "Stop Actor 1" would be very interesting.  For example if I had these two actor groups.

"recirculate htl" that opened valve 1, opened valve 2 and started motor 1  
"mash" that opens valve 1, opens valve 2 and starts motor 1 

Clearly both of these requires the other to NOT be running when it starts.  So it would be nice to be able to have a simple logic item "stop".  In the example above you could stop the other actor group before starting the new functions.

bertah, 2017-11-01:
While I agree that an hardware selector is much more secure, I am facing a similar issue and am currently working on a plugin that would let you create ExclusiveGPIO and ExclusiveGPIOPWM actors. Two actors belonging to the same "exclusive group" (can be set when creating the actor) would be prevented from turning on simultaneously. When I'm done with testing and publishing the plugin, I will let you know!

chrisbrucewilliams, 2017-11-01:
Thanks that would be cool.  One less item to put in the hardware box, which
is the main reason I started looking at the CBP.  I'd like to only have an
on/emergency stop button and maybe some indicator lights in the electric
box.

crazyzealot, 2019-03-10:
@bertah, has this been done now? I have the same problem and would very much like to be able to switch off my HLT element while RIMS are on

crazyzealot, 2019-03-10:
> While I agree that an hardware selector is much more secure, I am facing a similar issue and am currently working on a plugin that would let you create ExclusiveGPIO and ExclusiveGPIOPWM actors. Two [...]

@bertah, has this been done now? I have the same problem and would very much like to be able to switch off my HLT element while RIMS are on