Manuel83/craftbeerpi3 #153: Adjust power to heater

chrisbrucewilliams, 2017-11-01:
Using PID based solutions like http://www.theelectricbrewery.com/ it is possible to adjust the power to a heater through the PID.  How they do this I am not 100% sure of but I believe they are NOT adjusting the    volts send to the heater.  My understanding is they are doing something with the on/off cycles.  My first question is how are they doing this?

My second question is how is this done in craftbeerpi?  I see you can adjust the power going to a heater but is there anything else needed to actually get the power level of the heater to change?

HotPepperDaddy, 2017-11-01:
The Auber PID uses a duty cycle.  Even in manual mode it used the t setting (duty cycle time).  If it's set to 80% in manual mode and t is 2 seconds it will be on for 1.6s then off for 0.4s.  I don't know if anyone has implemented this in CBP yet but if no one beats me to it, I will.  My panel uses an Auber PID but I'm switching to CBP.

chrisbrucewilliams, 2017-11-01:
There is a selection in CBP to adjust the power, but I have no idea if you need additional hardware or if works like you said on duty cycle.  My understanding which is limited on duty cycle is it is a bit more complex than just doing 80% of the max.  I believe it does some sort of on/off cycling.  If CBP did the same without any hardware then problem solved.  Did you see my other post https://github.com/Manuel83/craftbeerpi3/issues/152 on multiple heaters running at once?  This is another heating element issue I am wondering about.

bertah, 2017-11-01:
I can confirm that you can do this in cbpi3 without additionnal hardware. When adding your GPIOPWM actor, you can set the duty cycle of the PWM as a parameter. Since the duty cycle is in Hz (cycles/second), just set the value as 1/x where x is your duty cycle in seconds. So to mimic the behavior of the Auber PID, which has a duty cycle of 2 sec, set the value to 0.5 (1/2 Hz). I am running mine at 0.1 which gives a 10sec duty cycle. So when setting the power at 80%, the element will stay on 8 seconds, off 2 seconds, and so on.

HotPepperDaddy, 2017-11-01:
For #152 a selector switch is safe, and basic. However, technically you could write the software so that two elements could be "on" at the same time as long as their "on" part of each duty cycle doesn't overlap. In that case you don't want any bugs causing a meltdown.

chrisbrucewilliams, 2017-11-01:
Bertah if you put it in the hardware config like you said this means you cannot edit it during brewing, correct?  How do you change it during brewing?

bertah, 2017-11-01:
Sorry, I mixed up the terminology, I edited my post above. The duty cycle is actually controlled by the "power" slider, that you can change during brewing. While the  frequency (the total "period") is fixed and set in the hardware config. So if your frequency is, say, 10 seconds (0.1), and you set the power to 80%, element will stay on 8 seconds, then off 2 seconds, and so on.

chrisbrucewilliams, 2017-11-02:
Cool thanks

lalo-uy, 2017-11-06:
There are 2 concepts here,  the power setting slider, and the PWM action of
the PDI.
If I set the slider at 80% then the 100% of the PDI should be that 80%.
Did I make myself clear?

linuxguy123, 2019-02-13:
Caution: some SSRs are not built to be cycled on/off at a high frequency. Remember, the temp sensors in CBPi3 are read every 5 seconds and most kettle heat systems won't raise the temp faster than a few F per minute. A PWM frequency of a second or more may be adequate and work well with most SSRs.

I like that someone is running the PWM on a 0.1 Hz (10 second) duration.

nkaramolegos, 2020-10-27:
So guys there is no voltage adjust. The PID logic creates a smart ON/OFF sequence. Correct? Also, the 80% of power means that the Duty cycle is 80%. Correct? Last question: There are the following choices for GPIO, GPIOSimple, GPIOPWM, GPIOrelay? Where should I use each one? If I need PID for turn on/off a electronic valve can I use GPIOrelay to turn on/off the valves? (the current is smaller than 10A so I don't need a SSR).

lalo-uy, 2020-10-28:
GPIOsimple is straight as you set on is 5V in pin / off is 0V on pin
GPIORelay is inverted logic    on is 0V / off is 5V
GPIOPWM allow you to set up the frequency and duty cycle. There used to be
a problem with the standar one, but there is a plugin that works ok.
For PWM you almost always need an SSR, unless the frequency is very very
low. The mechanical relay is not good to turn on and off once per second.

nkaramolegos, 2020-10-28:
Thank you. Nice details. Are you available via mail (or chat) for some more questions?