Manuel83/craftbeerpi3 #154: Touch screen

chrisbrucewilliams, 2017-11-01:
I'm planning to add a touch screen to my setup.  Of course I could use the pi lcd which goes thru its own connection, but I was wondering if anyone has done this using an off the shelf touch screen and the HDMI port?

hauca, 2017-11-02:
I run a Sceptre E165W-1600HC monitor with a AirBar by Neonode 15.6". Both plug right into RPI no drivers or setup needed.

HooTsBrooklyn, 2017-12-08:
I use a kuman 5 inch Resistive Touch Screen (https://smile.amazon.com/gp/product/B01F3EKJIA) on my fermentation chamber.  I have it plugged into the HDMI for video, USB for power, and I had to use some jumper cables from the back terminals to the GPIO pins for the touch capability.  I had to muck about with it for a bit to get it to work, but it looks and works great now.