Manuel83/craftbeerpi3 #155: Send SMS when step is complete

chrisbrucewilliams, 2017-11-05:
I was wondering how you could engage Amazon SNS on "step complete"?  If I remember correct when a step is complete GPIO 16 gets a 3V burst.  So I'm thinking I can add in the Amazon call in this code?  Or maybe it would be better if there was a plugin for this?  I'm am very far from a coder so detailed help would be great.  Thanks in advance.  

I plan on having this https://www.adafruit.com/product/2994 on my control box with a motion sensor to turn it off.  But I'd like to also like to have a SMS message sent.  

I was looking at this topic https://learn.adafruit.com/sms-texting-pet-food-dish/software?view=all on how to send a message when your pet food bowl is empty.  To do this they used Amazon Simple Notification Service https://aws.amazon.com/sns/ which is free for a small number of text messages a month.

Philusha1983, 2017-11-06:
I think your option is overkill.

you can use the pusher plugin and get the same results as a push
notification.
no additional hardware.
ᐧ

chrisbrucewilliams, 2017-11-06:
Where is the best place to get a description of all the plugins?  Also is there instructions on how plugins are built?

Philusha1983, 2017-11-06:
In the plugin section you have links to all the github pages

chrisbrucewilliams, 2017-11-06:
Sorry to be such a newbie.  Where is the "plugin section"?

Philusha1983, 2017-11-06:
if you instal the craft beer pi 3 in the menu you have plugins\ addons.
you get a list of availble plugins and 

*Philip* :)

Philusha1983, 2017-11-06:
if you instal the craft beer pi 3 in the menu you have plugins\ addons.
you get a list of availble plugins and links to the github pages.

*Philip* :)