Manuel83/craftbeerpi3 #159: Create a task step and reminders during steps

ksullivan86, 2017-12-04:
It would be great if some how while making the brew list you could add steps that discribe an action to be preformed in a text box and a done button to move on to the next step. Examples that this could be used for could be; opening a manual valve, installing a sparge arm, taking a sample... so many things that can sometime be forgotten if you are distracted. Also not sure if this would be a separate item but it would also be nice to have reminders in the boil step or other steps that flashed a on screen text box reminding you of a custom step could time controlled : put chiller in hot wort to sanitize at 15min , sanitize fv, oxygenate... anything that can be forgotten but. It could also be useful if the reminder could be a list that you could check off as you go. I guess the difference is a step would stop a timer while a reminder wouldn’t. Just a thought it would really help me during my brew day not to have to ask myself what’s next. Sorry if this isn’t entirely Coherent as this idea has grown matured while typing this out

Philusha1983, 2017-12-04:
Great idea, I favour it

lalo-uy, 2017-12-04:
Great, if you can save the recipe.

Saving & loading recipes is far more important.