Manuel83/craftbeerpi3 #160: Option for custom background

KenCope, 2017-12-07:
I have attempted to write the code to change the background using the https://github.com/Manuel83/craftbeerpi3/blob/master/modules/ui/static/index.html file.  No success for me.  Can anyone help?  Manuel suggested some code for me, no success either.  Thanks in advance!

**Manuel suggested:

**_You need to add a custom css in this file.

https://github.com/Manuel83/craftbeerpi3/blob/master/modules/ui/static/index.html

Please add the following code in line 13

<style>
    body {
        background-image: url("static/bg.jpg") !important;
    }

</style>

The background image need to be placed in modules/ui/static/bg.jpg_****

HooTsBrooklyn, 2017-12-08:
If you just add background-image:url('bg,jpg'); under the body declaration in the bootstrap.dark.css file, put the bg.jpg file in the ui/static/ folder and it'll work.
[image]
[image]
[image]

I un-minimized the CSS file for ease of reading though

KenCope, 2017-12-08:
For some reason I can't get this to work.  I open the bootstrap.dark.css file using text editor, insert the verbiage you gave me and all it does is prints that on the screen in the background.  I appreciate everyone's help.

KenCope, 2017-12-08:
[Image]

Is the file size of the.jpg limited?

[virtualbox_raspberrypi clone_07_12_2017_18_26_45]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F4926059%2F33745403-4da22a9a-db7c-11e7-88cf-ab925989653b.png&data=02%7C01%7C%7C51623e1c5e854256665708d53dd2e5f5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636482898358755420&sdata=bQnX9Z6Q2P2uEkERB4Dzt2NWWalHOyPZEKasqnrd25c%3D&reserved=0>
[virtualbox_raspberrypi clone_07_12_2017_18_27_04]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F4926059%2F33745404-4dc5d850-db7c-11e7-8860-f1ac15b9437d.png&data=02%7C01%7C%7C51623e1c5e854256665708d53dd2e5f5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636482898358755420&sdata=4vPVzinIoBE6WMVJy7qiBHEK2Hv69fb%2FnPtcNDckOJE%3D&reserved=0>
[virtualbox_raspberrypi clone_07_12_2017_18_28_31]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F4926059%2F33745432-754b6822-db7c-11e7-9dce-eaff5130995e.png&data=02%7C01%7C%7C51623e1c5e854256665708d53dd2e5f5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636482898358755420&sdata=%2Fx0Zpf3Vv9bxbYWZov04toc2tkW3cUQQSLU1yXke4ww%3D&reserved=0>

HooTsBrooklyn, 2017-12-08:
Looks like I made a typo.  It should be bg.jpg not bg,jpg.  If you are using chrome, right click the background and select inspect element.  Select the console tab and see if there's an issue loading the image (error 404).

KenCope, 2017-12-08:
Failure to load. bg.jpg not found. But that file is in there. I’ve even tried a few different .jpg files.  Frustrated :-/

Ken[Image]

KenCope, 2017-12-08:
I’m an idiot!!  It can’t find bg.jpg when I have the file names bp.jpg!  Thanks to all for your help! I’ll consider this case closed. Happy holidays.

Ken

closed 2018-05-06 by KenCope