Manuel83/craftbeerpi3 #162: No entry in sensor list

fencerJP, 2017-12-09:
I'm installing cbpi3 for the first time, so I apologize if I've made a mistake somewhere. I THINK I have the one wire sensor wired up correctly (per http://web.craftbeerpi.com/hardware/ ), but I can't get the temperature to display in cbpi3.

On my main page, I get 0 deg:
[image]

While in my hardware setup, I get:
[image]

I expected something like this, from the setup video:
[image]

What am I doing wrong?

HooTsBrooklyn, 2017-12-09:
Have you rebooted the Pi with the sensor connected?  I noticed that mine didn't show up in the list until I had them connected when the CBPi service started.  Once I had them setup, I could connect/disconnect them at will.

fencerJP, 2017-12-23:
Well now I suspect something with my GPIO overall. Today I attempted to just turn on/off an SSR, and was unable to do so, despite testing two different SSRs, attaching the SSR to the 3v3 pin 1, and reconfiguring  the pins vs GPIO numbers (because pin 12 is GPIO18, but the settings were unclear which one number it was referring to). 

I will find a way to test the SSRs, but is it possible to have a faulty GPIO? Is there some way to test it besides using a multimeter? The Pi itself obviously works.

lalo-uy, 2017-12-26:
Som SSr do not triger with 3V.
Try them witn 5V.
Try the GPIO with a led + 150 ohms R

fencerJP, 2018-01-01:
Well after a re-install and a re-wire, I'm not sure what the issue was. Everything APPEARS to be working during a bench test, but we'll see how it goes when I put everything in the box.

closed 2018-01-01 by fencerJP