Manuel83/craftbeerpi3 #165: Craftbeerpi refused to start.

oleost, 2017-12-30:
I had a working setup using SSR for brewing and tp-link wireless for fermentation.

Next time I tried to make a brew after I had used fermentation it didn't work, without me having done anything. ( It refused connection on web interface )

SSH into the box, tried stop and start of Craftbeerpi, I also tried to run install again without that helping. Lastly I did clean the database and then it started working.

pi@raspberrypi:~/craftbeerpi3 $ sudo python run.py
INIT DB
/home/pi/craftbeerpi3/modules/buzzer/__init__.py:18: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(self.gpio, GPIO.OUT)
Traceback (most recent call last):
  File "run.py", line 3, in <module>
    from modules import socketio, app, cbpi
  File "/home/pi/craftbeerpi3/modules/__init__.py", line 64, in <module>
    cbpi.run_init()
  File "/home/pi/craftbeerpi3/modules/core/core.py", line 457, in run_init
    i.get("function")(self)
  File "/home/pi/craftbeerpi3/modules/plugins/TPLinkPlugin/__init__.py", line 163, in init
    TPplugs = getPlugs()
  File "/home/pi/craftbeerpi3/modules/plugins/TPLinkPlugin/__init__.py", line 150, in getPlugs
    deviceList = resp["result"]["deviceList"]
KeyError: 'result'
pi@raspberrypi:~/craftbeerpi3 $

mrillies, 2018-01-01:
Looks like the TPLink Plugin needs some better error checking.

mrillies, 2018-01-01:
Were you using the TPLink cloud setup?

Looks like the cloud API returned an error message or something. i have submitted a pull request on the TPlink git hub, once its accepted you should update the plugin

oleost, 2018-01-01:
@mrillies Yes I believe I was logged into to TP-link acount. 
Thanks for quick fix, hopefully it will fix the issue.

https://github.com/aravndal/TPLinkPlug/pull/3/commits/7824d4cb2a9c162f48c7327e0fb37fd9f51dc388

But ideally an Pluging should not hinder Craftbeerpi to start? If the plugin malfuncitons, craftbeerpi should simple disable it ?

mrillies, 2018-01-02:
> aravndal/TPLinkPlug@7824d4c

Agreed, I’ve raised this as an issue before and hopefully it gets fixed in
3.1.

I could make changes now to fix it but don’t want to mess with what Manuel
has planned for 3.1

oleost, 2018-03-16:
@mrillies @Manuel83 Looks like this have not been merged to plugin github by the owner.

Is it really benificall to keep plugins on seperate Githubs ?
This is fairly new already, and already there is "dead" plugins ?

Would it be better to keep plugins in master Github ?

archonic, 2018-04-05:
Plugins are made by other authors most of the time. If they were maintained all in one repo, that means just the admins for this repo would be responsible for their development and maintenance and would destroy most of the point of having a plugin system.