Manuel83/craftbeerpi3 #167: Raspbian Stretch, cannot reach Craftbeerpi3 GUI after installation

Torstein80, 2018-01-18:
Hello
I can't get Craftbeerpi3 to work on the new release of Raspbian Strecth. I get the message: "This site can't be reached" "myip" refused to connect.

I am able to install Craftbeerpi3, but for some reason i cannot reach the web interface. I tried the following: http://myipadress:5000 and http://localhost:5000. I did add Craftbeer to autostart, tried to re-launch the install.sh file and chose stop, clear and restart craftbeerpi, and i have of course rebootet several times.

Anyone else having this issue? I have two perfectly working installs on Raspbian Pixel. I guess I miss something obvious, just cant see it, yet...

HooTsBrooklyn, 2018-01-18:
It's a known bug.  Try following the instructions here
https://cbpi.mricesolutions.com/viewtopic.php?f=3&p=14

Torstein80, 2018-01-19:
Thank you  @HooTsBrooklyn, adding the missing dependencies worked like a charm.
For info, I added the following dependencies for this Raspbian Stretch release to make Craftbeerpi3 to work:
sudo pip install flask_socketio
sudo pip install gitpython
sudo apt-get install python-yaml

Again, thank you.

closed 2018-01-21 by Torstein80