Manuel83/craftbeerpi3 #168: CBPi3 Plugins - Where'd they go?

KenCope, 2018-01-20:
I just reinstalled CBPi and I see the plugin __init__.py file is empty.  Is there a fix or something I'm missing?  The plugins always worked for me before.  Thanks.

HooTsBrooklyn, 2018-01-20:
Looks like the plugins repository broke yesterday.
https://github.com/Manuel83/craftbeerpi-plugins

ksullivan86, 2018-03-12:
Is this still an issue? I did a fresh install and now I don't see any plugins load when I go to the plugins page not sure what to do, any ideas?

cpompermaier, 2018-03-14:
Same problem here. I just installed and had no errors during installation.

HooTsBrooklyn, 2018-03-14:
It should be good now.  One of the added plugins had a small issue that was fixed.

cpompermaier, 2018-03-14:
Thank you. Working good now.

ckrack, 2018-04-05:
@KenCope please close the issue.

closed 2018-04-05 by KenCope